bolderflight/ublox #3: Multiple Receivers

alorman, 2018-04-04:
Brian;
Awesome software libraries, much thanks!
Is there an easy way to call the library to talk to two ublox units on two different serial ports?
I can't figure this out, but programming also isn't my exact forte.
Much thanks!

flybrianfly, 2018-04-04:
Should work fine be declaring multiple uBlox objects, like:

```C++
#include "UBLOX.h"

UBLOX gps1(1);
UBLOX gps2(2);

gpsData uBloxData1;
gpsData uBloxData2;

void setup() {
  Serial.begin(9600);
  gps1.begin(115200);
  gps2.begin(115200);
}

void loop() {
  if( gps1.read(&uBloxData1) ){
    Serial.print("GPS1 Data:");
    Serial.print("\t");
    Serial.print(uBloxData1.utcYear);  ///< [year], Year (UTC)
    Serial.print("\t");
    Serial.print(uBloxData1.utcMonth); ///< [month], Month, range 1..12 (UTC)
    Serial.print("\t");
    Serial.print(uBloxData1.utcDay);   ///< [day], Day of month, range 1..31 (UTC)
    Serial.print("\t");
    Serial.print(uBloxData1.utcHour);  ///< [hour], Hour of day, range 0..23 (UTC)
    Serial.print("\t");
    Serial.print(uBloxData1.utcMin);   ///< [min], Minute of hour, range 0..59 (UTC)
    Serial.print("\t");
    Serial.print(uBloxData1.utcSec);   ///< [s], Seconds of minute, range 0..60 (UTC)
    Serial.print("\t");
    Serial.print(uBloxData1.numSV);    ///< [ND], Number of satellites used in Nav Solution
    Serial.print("\t");
    Serial.print(uBloxData1.lat,10);   ///< [deg], Latitude
    Serial.print("\t");
    Serial.print(uBloxData1.lon,10);   ///< [deg], Longitude
    Serial.print("\t");
    Serial.println(uBloxData1.hMSL);   ///< [m], Height above mean sea level
  }

  if( gps2.read(&uBloxData2) ){
    Serial.print("GPS2 Data:");
    Serial.print("\t");
    Serial.print(uBloxData2.utcYear);  ///< [year], Year (UTC)
    Serial.print("\t");
    Serial.print(uBloxData2.utcMonth); ///< [month], Month, range 1..12 (UTC)
    Serial.print("\t");
    Serial.print(uBloxData2.utcDay);   ///< [day], Day of month, range 1..31 (UTC)
    Serial.print("\t");
    Serial.print(uBloxData2.utcHour);  ///< [hour], Hour of day, range 0..23 (UTC)
    Serial.print("\t");
    Serial.print(uBloxData2.utcMin);   ///< [min], Minute of hour, range 0..59 (UTC)
    Serial.print("\t");
    Serial.print(uBloxData2.utcSec);   ///< [s], Seconds of minute, range 0..60 (UTC)
    Serial.print("\t");
    Serial.print(uBloxData2.numSV);    ///< [ND], Number of satellites used in Nav Solution
    Serial.print("\t");
    Serial.print(uBloxData2.lat,10);   ///< [deg], Latitude
    Serial.print("\t");
    Serial.print(uBloxData2.lon,10);   ///< [deg], Longitude
    Serial.print("\t");
    Serial.println(uBloxData2.hMSL);   ///< [m], Height above mean sea level
  }
}

```

alorman, 2018-04-04:
Awesome. You're a star!
Thanks!

closed 2018-04-04 by alorman