bolderflight/ublox #4: Wake from sleep function

jacky4566, 2018-10-17:
Could you program a clear fix function?

When waking a microprocessor and the Ublox from deep sleep (no millis) there is no way to determine if a fix is new or old. It would be easiest to clear the data before waking the GPS.

I may fork this and write the function myself. Stay tuned!

flybrianfly, 2018-10-17:
Interesting suggestion. Probably would want to clear all of the data since it would no longer be valid. Let me think about the best API interface for this, should be easy enough to code.

jacky4566, 2019-05-06:
In my latest pull i have tweaked the begin function to clear out the latest packet. thusly when begin is called the library has no data. Fresh.