bolderflight/ublox #9: SoftwareSerial...

kugelkopf123, 2019-07-03:
Hi! 

Is there a possibility to use it with a SoftwareSerial. I'am using a ESP8266 so there is no HWSerial. 
I've already tried to change some lines in UBLOX.h and UBLOX.cpp but i don't get it. 
I really need some help with this. Thanks...

flybrianfly, 2019-07-03:
Seems like you just need to modify the source and header to use a software serial object instead of a hardware serial object. Lines 52 and 111 in UBLOX.h and line 28 in UBLOX.cpp

kugelkopf123, 2019-07-03:
Yeah. It could be so easy sometimes... 
Now i have no Errors anymore while its uploading but i don't get any results. I use the UBLOX_Example.ino. But there are no results to print. My Neo7m Module is connected correctly.  I've tried it via Serialbridge. 
As far as i can see, my "gps.readSensor()" is always false. My GPS Module is 3dFixed, I've checked it via u-Center. 

In: UBLOX.cpp `bool UBLOX::_parse(uint8_t msg_class,uint8_t msg_id,uint16_t msg_length)`
I get some Stuff if I print the result of var "_byte". : 

```
bool UBLOX::_parse(uint8_t msg_class,uint8_t msg_id,uint16_t msg_length)
{
	// read a byte from the serial port
	Serial.print("AVA: ");Serial.println(_bus->available());
	while (_bus->available()) {
			
		_byte = _bus->read();
		Serial.print("byte: ");Serial.println(_byte );
		
		// identify the packet header
		if (_parserState < 2) {
			if (_byte == _ubxHeader[_parserState]) {
				_parserState++;
			} else {
				_parserState = 0;
			}
		} else {
			if ((_parserState - 2) < msg_length) {
				*((uint8_t *) &_tempPacket + _parserState - 2) = _byte;
			}
			_parserState++;
			// compute checksum
			if ((_parserState - 2) == msg_length) {
				_calcChecksum(_checksum,((uint8_t *) &_tempPacket),msg_length);
			} else if ((_parserState - 2) == (msg_length + 1)) {
				if (_byte != _checksum[0]) {
					_parserState = 0;
				}
			} else if ((_parserState - 2) == (msg_length + 2)) {
				_parserState = 0;
				if (_byte == _checksum[1]) {
					memcpy(&_validPacket,&_tempPacket,sizeof(_validPacket));
					return true;
				}
			} else if (_parserState > (msg_length + 4) ) {
				_parserState = 0;
			}
		}
	}
	return false;
}

```

```
AVA: 63
byte: 0
byte: 181
byte: 98
byte: 1
byte: 7
byte: 84
byte: 0
byte: 232
byte: 89
byte: 205
byte: 19
byte: 227
byte: 7
byte: 7
byte: 3
byte: 20
byte: 16
byte: 47
byte: 7
byte: 16
byte: 0
byte: 0
byte: 0
byte: 236
byte: 249
byte: 255
byte: 255
byte: 3
byte: 1
byte: 10
byte: 6
byte: 34
byte: 39
byte: 51
byte: 4
byte: 250
byte: 77
byte: 30
byte: 31
byte: 240
byte: 69
byte: 1
byte: 0
byte: 255
byte: 145
byte: 0
byte: 0
byte: 9
byte: 16
byte: 0
byte: 0
byte: 54
byte: 20
byte: 0
byte: 0
byte: 248
byte: 255
byte: 255
byte: 255
byte: 245
byte: 255
byte: 255
byte: 255

AVA: 63
byte: 181
byte: 98
byte: 1
byte: 7
byte: 84
byte: 0
byte: 208
byte: 93
byte: 205
byte: 19
byte: 227
byte: 7
byte: 7
byte: 3
byte: 20
byte: 16
byte: 48
byte: 7
byte: 15
byte: 0
byte: 0
byte: 0
byte: 7
byte: 250
byte: 255
byte: 255
byte: 3
byte: 1
byte: 10
byte: 7
byte: 27
byte: 39
byte: 51
byte: 4
byte: 249
byte: 77
byte: 30
byte: 31
byte: 138
byte: 70
byte: 1
byte: 0
byte: 153
byte: 146
byte: 0
byte: 0
byte: 8
byte: 16
byte: 0
byte: 0
byte: 47
byte: 20
byte: 0
byte: 0
byte: 248
byte: 255
byte: 255
byte: 255
byte: 245
byte: 255
byte: 255
byte: 255
byte: 14

```
In uCenter I have activated NAV-PVT for UART1. And on UBX-CFG-PRT: UART1 : UBX

I have no Idea whats the problem. Tried different Baudrates from 9600 to 115200.

flybrianfly, 2019-07-03:
Could it be this the same as issue #1? Since you are using a uBlox 7.
https://github.com/bolderflight/UBLOX/issues/1

kugelkopf123, 2019-07-03:
Thanks, i had seen this issue and tried it, but that was not the problem. I think the problem is the Buffersize of the SoftwareSerial. It is only 64Byte, but the Payload is greater. So i have to find a way to deal with this... If i found something I will post it here.

kugelkopf123, 2019-07-04:
Ok, i have solution for it. But first: How can I block a certain function until its over? 
If the gps.readSensor() function gets called in the loop() then everytime its called it makes a new "instance" of this function with the same Serial input for each one. 
Example: If the loop() calls gps.readSensor() every 100ms but the readSensor function needs about 1000ms than its for nothing.

I simply want that the gps.ReadSensor function blocks multiply calls. 

Any Ideas how to do this inside the UBLOX.h File?

kugelkopf123, 2019-07-09:
Ok. Just for now. I switched to an ESP32 because i hadnt enough IOs on my ESP8266. So this isnt my Problem anymore. Further i started to add some functions into this Library to Send UBX Data to my Neo7. 
To help other People that have Problems with their ESP8266, this was my solution: 
As Mentioned here: 

> Seems like you just need to modify the source and header to use a software serial object instead of a hardware serial object. Lines 52 and 111 in UBLOX.h and line 28 in UBLOX.cpp

A low selected Baudrate and:
`SoftwareSerial ss(D1,D2 false, 256);`

Set the Buffer on 256.

Setup the NEO7 In/Out (UBX-CFG-PRT) Port to UBX In/Out only for UARt1 and in UBX-CFG-MSG only activate NAV PVT for UART1. Ohterwise the Buffer gets really messed up with shit that cant be interpreted.  
Have Fun!

closed 2019-07-09 by kugelkopf123