bolderflight/ublox #16: 'elapsedMillis' does not name a type

boris-gu, 2022-09-01:
I have:  
ESP32 DevKit v1  
ubx v5.1.0  
units v4.1.1  
eigen v3.0.2  

When I compile ublox_example.ino, I get an error  
```
In file included from /home/boris/.var/app/cc.arduino.IDE2/cache/.arduinoIDE-unsaved202281-31-1aa1z6a.w72z/ublox_example/ublox_example.ino:26:0:
/home/boris/Документы/Projects/Arduino/libraries/ublox-5.1.0/src/ubx.h:161:3: error: 'elapsedMillis' does not name a type
   elapsedMillis t_ms_;
   ^

exit status 1

Compilation error: exit status 1
```

boris-gu, 2022-09-06:
I have included [elapsedMillis library](https://github.com/pfeerick/elapsedMillis) in ubx.h and now everything works.

I do not know if this error appeared due to the peculiarity of using ESP32 or something else.

flybrianfly, 2022-10-04:
Thanks, I fixed this in the latest release.

closed 2022-10-04 by flybrianfly