bolderflight/ublox #17: NEO-M8N module without error

rouhUAV, 2022-10-03:
Dear developers
I have tried your ubx lib with a NEO-M8N module but i couldn't make use of it. I have test my hardware with some other libraries like sparkfun and it worked, as I like to use this ublox lib with some of the other libraries that you have provided, could you please inform me whether its a version issue? whats your suggestion? shall I update my GPS firmware?(currently its protocol version is 15.00). 
The fact is that it happens without any error, the program compiles with my arduino 2 IDE and I can program my ESP32 board successfully. the program runs until the myGNSS.Begin() is called.

flybrianfly, 2022-10-04:
Thanks. Please try with the latest version of this library (v6.0.3) and the latest firmware with the NEO-M8N. Also check that you have all the required messages enabled and have set the correct serial port and baudrate. I just tested this with a very old NEO-M8N that I have - I wasn't able to enable the EOE message in u-center until after I updated firmware. Then I was able to enable all of the required messages and it worked with this library fine.

rouhUAV, 2022-10-04:
Thank you @flybrianfly, I have checked the baudrate with u-center. but I am not deeply sure about the serialPort, the ESP32 serial port in arduino is a bit vague, 

The software version is 2.01(75350) I hope it would be sufficient because my NEO-M8N module is probably not updatable.
I will try your suggestions.
Thank you very much Brian.