bolderflight/ublox #19: the member named pi does not good idea

EeeLo, 2023-08-09:
thanks for the best lib befor, and I'm using this lib to with other project and the error is "pi", I think this is not good idea to using pi in this code.

flybrianfly, 2023-08-09:
To the best of my knowledge, there is nothing in the library using "pi". Can you please ensure that you're using the latest library downloaded directly from GitHub and provide a screenshot of the error?

EeeLo, 2023-08-12:
Hi,I'm using "sparkfun sgp4 arduino library" to making a tracking board,this lib using pi,and I think the pi name always in the space calibrtion code,thanks for your good library.

flybrianfly, 2023-08-13:
Sounds like the issue is not with this library, so I'm closing it.

closed 2023-08-13 by flybrianfly